bythorsthunder/Voron_Mods #10: Wristwatch G2 Dual Filament Sensor: Unreliable Filament detection

KlausHans, 2025-03-17:
Hello thanks for the filament sensor remix of the Wristwatch G2. I tried it and for it is not reliable for both sensors. They get triggered and untriggered if i wiggle the filament. This also happens with printhead movements. 

I am using 5.5 mm steel balls.

bythorsthunder, 2025-03-17:
Thank you for the note Klaus.  I have been looking at ways to improve the reliability of the detection and some other users have also created some modifications of the original extruder to that end.  Are you on the Voron discord server?  If so you can ping me there and I can point you to the modified files to try out.

KlausHans, 2025-03-17:
Thank you for the quick response and help. I am on Discord but i kind of hate it. So much good knowledge and help is hidden there, unreachable for search engines. 
I will ping you sometime, i need some sleep now. :)